MicrosoftDocs/windows-driver-docs #426: where the virtual address came from

diyanyagari, 2018-04-12:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a5e088bd-7daa-d560-1c97-5f4e10009238
* Version Independent ID: ea3f733e-3906-f445-213c-8201c38284f9
* Content: [Converting Virtual Addresses to Physical Addresses](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/converting-virtual-addresses-to-physical-addresses)
* Content Source: [windows-driver-docs-pr/debugger/converting-virtual-addresses-to-physical-addresses.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/converting-virtual-addresses-to-physical-addresses.md)
* Service: **unspecified**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

HEY : 
MyApp.exe process is running and you want to investigate the virtual address 0x0012F980

why you not explain where the virtual address came from

EliotSeattle, 2018-04-16:
@DOMARS Maybe a link to discussion about virtual addresses and how to get them?

DOMARS, 2018-05-07:
@diyanyagari - Thanks for the question on virtual address information. I added a link to the Virtual address spaces -- https://docs.microsoft.com/en-us/windows-hardware/drivers/gettingstarted/virtual-address-spaces. I also recently added a few general resources, to the getting started topic, that might be helpful, including the classic Windows Internals book that has a great discussion of Windows memory. 
https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/getting-started-with-windows-debugging.

closed 2018-05-07 by DOMARS