MicrosoftDocs/windows-driver-docs #429: Driver promotion properties of New Shipping Label page.

armstrong-charles, 2018-04-13:
As in the documentation, there are three driver promotion options available. 
It is not a issue, it is a question about this option - Automatically deliver and install this driver on all applicable system to promote your driver to Critical.
When i select this option for my driver update, does it reflect in the 'Classification"  column as a Critical Update in the Windows Update Catalog.
I have seen various drivers that are automatically downloaded and installed, but those drivers are not marked as Critical update in the Classification column of the Windows Update Catalog.
Your insights will be appreciated. 
Thanks

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 531ef6ee-7a29-e6d0-56d6-e08a923b0aa4
* Version Independent ID: d00f4b65-2ae6-643f-d67a-c04f916a7058
* Content: [Publish a driver to Windows Update](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/publish-a-driver-to-windows-update)
* Content Source: [windows-driver-docs-pr/dashboard/publish-a-driver-to-windows-update.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/publish-a-driver-to-windows-update.md)
* Service: **unspecified**
* GitHub Login: @aahill
* Microsoft Alias: **windowsdriverdev**

aahill, 2018-04-18:
Hi @armstrong-charles, I looked into this and the driver metadata that determines if a driver is optional or critical isn't exposed in the Windows Update Catalog. Additionally, a driver's classification doesn't determine whether it's optional or critical - the classification is always "Drivers". The "Critical updates" classification is only relevant in regards to software updates. Hopefully that helps?

armstrong-charles, 2018-04-19:
Hi @aahill thanks, it really helps. This is my understanding that 'Critical Updates' classification is only relevant in regards to software updates and classification of Driver update is always "Drivers" that comes from the metadata of the package, however, driver promotion option promotes the driver delivery, allowing how driver can be delivered to Windows Clients.

aahill, 2018-04-24:
Glad to help, thanks for reaching out!

closed 2018-04-24 by aahill