MicrosoftDocs/windows-driver-docs #430: Missing description for parameter 0x61948

StephenCleary, 2018-04-14:
I've gotten this one.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: de955911-dfbf-33b3-778a-767a5a578a0d
* Version Independent ID: 23fb5cbd-5f7d-f7ca-1622-5885020966c2
* Content: [Bug Check 0x1A MEMORY_MANAGEMENT](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x1a--memory-management)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md)
* Service: **unspecified**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2018-04-17:
@StephenCleary - Thanks for pointing this out. I will investigate with the dev team and will document if applicable.

closed 2018-04-17 by DOMARS

KylePhoria, 2018-12-05:
I've got the same bug check with parameter 0x61948.
There is still no update until now.