MicrosoftDocs/windows-driver-docs #432: Debuggee is running...

serup, 2018-04-16:
Debugger somehow never gets into a state where I can actually break all into it.
It stays there after "Task "Driver Post Install Actions" completed successfully

Pressing Break all - in the debug menu does nothing !!!

Any suggestions to what could be wrong ? I just made a standard wizard kmdf driver and followed above changes

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 81568a12-c39e-a18f-8f33-c54545f14bac
* Version Independent ID: 73217b22-1795-b5d0-65cb-1984f38dc661
* Content: [Write a universal Hello World driver (KMDF)](https://docs.microsoft.com/en-us/windows-hardware/drivers/gettingstarted/writing-a-very-small-kmdf--driver)
* Content Source: [windows-driver-docs-pr/gettingstarted/writing-a-very-small-kmdf--driver.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/gettingstarted/writing-a-very-small-kmdf--driver.md)
* Service: **unspecified**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

duncanmacmichael, 2018-04-30:
Hi @serup , sorry you're having this issue with trying to debug the sample driver. To clarify, you deployed the driver to a test computer, then, on the host computer, tried to hit F5 to debug within Visual Studio? Knowing how you got to where you are can help us figure out what steps in the documentation are unclear. Thanks!

duncanmacmichael, 2018-05-02:
Hi again @serup , we've updated the steps for deploying and debugging the example driver for this topic to reflect the recommended debugging method. The changes should be live soon. Let us know if this helps you break into the debugger and experiment with the commands. 

If you're still having issues, we recommend posting your question on the following forums, where other driver developers and Microsoft engineers can be found. The community there is very responsive.

[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
[OSR's ntdev List](http://www.osronline.com/showlists.cfm?list=ntdev)

Thanks for the feedback in improving the docs!

closed 2018-05-02 by duncanmacmichael