MicrosoftDocs/windows-driver-docs #438: Venu

vbusireddy, 2018-04-18:
In the sentence "The NetVSC exposes a Virtual Machine (VM) network adapter which is bound to the PF miniport driver to support the VF data path", shouldn't the "PF miniport" be "VF miniport", because the NetVSC is binding to the VF miniport driver?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0af351eb-8d84-6796-5916-853cc396ff31
* Version Independent ID: bc495f09-1bd9-45b7-acea-726f7b58703d
* Content: [SR-IOV VF Failover and Live Migration Support](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/sr-iov-vf-failover-and-live-migration-support)
* Content Source: [windows-driver-docs-pr/network/sr-iov-vf-failover-and-live-migration-support.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/sr-iov-vf-failover-and-live-migration-support.md)
* Service: **unspecified**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

duncanmacmichael, 2018-04-18:
Hi @vbusireddy , that could very well be the case but I'm not 100% sure. 

By saying that the "NetVSC is bound to the PF miniport driver to support the VF data path," it might be talking about the left path in the diagram, meaning the PF miniport driver binding is there _in case the VF path needs to fail over._ If this is true, then the sentence is correct but it could be clarified. The subsequent sentences talking about removing the VF miniport driver seem to corroborate this.

If the sentence is supposed to be talking about the right path in the diagram, meaning it's just describing the VF data path, then yes it is a typo. Let me see what I can find out for you.

duncanmacmichael, 2018-04-18:
Hi @vbusireddy , I found out it is indeed a typo. :) I'll go ahead and make the change; it should be live shortly. Thanks for the catch!

closed 2018-04-18 by duncanmacmichael

vbusireddy, 2018-04-18:
Thanks for taking care of this so expeditiously! :-)