MicrosoftDocs/windows-driver-docs #439: Issues about "_DSM interface for Byte Addressable Energy Backed Function Class"

MKingSWang, 2018-04-19:
Section "Storage Class Memory" described some DSM method to manage NVDIMM - N, is this section a design guide or a user guide?

EliotSeattle, 2018-04-19:
Hi @MKingSWang. 
Thanks for reaching out. Can you explain a little more? Is this a matter of nomenclature, or is there some issue with the content itself. 
Thanks,
Eliot

aahill, 2018-06-18:
Hi @MKingSWang, sorry for the delay, I'm taking over this documentation for Andrew temporarily .  You're referring to [this](https://docs.microsoft.com/windows-hardware/drivers/storage/-dsm-interface-for-byte-addressable-energy-backed-function-class--function-interface-1-) documentation, am I correct?

I'm not terribly familiar with this area, but it seems that these are implemented methods. You can find some more info about _DSM here as well: https://docs.microsoft.com/windows-hardware/drivers/bringup/acpi-device-specific-methods

Thanks!

closed 2018-06-18 by aahill