MicrosoftDocs/windows-driver-docs #443: AdvancedINF ?

redknight99, 2018-04-23:
Hi, I'm currently dealing with a .INF file with the Version key "AdvancedINF" . The full entry is "AdvancedINF=2.5". If anyone reads the feedback. I'd really appreciate any documentation on that key (What it is, What Windows versions it affects, etc) . Thank you. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 89e65b9c-b342-6df7-e878-4b88a9590363
* Version Independent ID: 8f0d7eb7-cd13-2769-f20d-296ecc96d417
* Content: [INF Version Section](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/inf-version-section#feedback)
* Content Source: [windows-driver-docs-pr/install/inf-version-section.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/inf-version-section.md)
* Service: **unspecified**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

redknight99, 2018-04-24:
Hi @tedhudek sorry for my vague initial question. To hopefully help clarify things. I'm currently working with an .INF file that has a version header that looks like. 

[version]
Signature=$chicago$
AdvancedINF=2.5

I'm trying to find out what the AdvancedINF=2.5 entry meant. I looked at the official version documentation page https://docs.microsoft.com/en-us/windows-hardware/drivers/install/inf-version-section but nothing's listed there about it. Can you lend a hand Ted?

tedhudek, 2018-04-24:
Hi @redknight99 thanks for the question.  I actually did ask around and no one seems familiar with it.  There was mention of it in old Internet Explorer code - could it have been related to that?  Not something we currently support though.

closed 2018-04-24 by tedhudek