MicrosoftDocs/windows-driver-docs #445: Ambiguous rules for 64-bit driver on Windows 10 with Secure Boot enabled

fpl9000, 2018-04-23:
Hello,

This page is ambiguous about the signing requirements in a particular use-case: installing a 64-bit driver on Windows 10 with Secure Boot enabled. At the top of the page, we see this:

"Starting with new installations of Windows 10, version 1607, Windows will not load any new kernel mode drivers which are not signed by the Dev Portal."

but with the exception that:

"Cross-signed drivers are still permitted if any of the following are true:
- ...
- Secure Boot is off."

But in the section titled "Signing a driver for earlier versions of Windows", we see a table that says when Secure Boot is enabled, a 64-bit driver must have a WHQL signature, which requires passing HLK/HCK tests.

It's not clear if this table only applies to Windows 7 and 8 — or to Windows 7, 8, and 10. If the former, then installing a 64-bit driver on Windows 10 with Secure Boot enabled requires attestation-signing the driver. If the latter, then installing a 64-bit driver on Windows 10 with Secure Boot enabled requires a WHQL signature. These are very different requirements.

Thanks for your attention to this!
--
Fran Litterio

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1eb19d33-8f4c-f379-b645-938a746c5615
* Version Independent ID: 8efb9fb4-1e9a-1a8a-994b-450495449e15
* Content: [Driver Signing Policy](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/kernel-mode-code-signing-policy--windows-vista-and-later-)
* Content Source: [windows-driver-docs-pr/install/kernel-mode-code-signing-policy--windows-vista-and-later-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/kernel-mode-code-signing-policy--windows-vista-and-later-.md)
* Service: **unspecified**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

fpl9000, 2018-04-24:
FYI, see this thread on this issue on the NTDEV mailing list. Peter Viscarola states that the table is in error when it says that a WHQL signature is required for a 64-bit driver on Windows 8.1 with Secure Boot enabled.
--
Fran Litterio

tedhudek, 2018-04-26:
I have some updates for this that I will be adding soon.  Thanks @fpl9000 for bringing it to our attention.  Look for an update in the next couple days.

tedhudek, 2018-04-27:
Note that @timrprobocom opened #451 to update the docs for this one.  Will update this issue once we get that one all set

fpl9000, 2018-04-28:
Thanks, Ted. I would defer to @timrprobocom's input on this topic. He is a regular contributor to the NTDEV mailing list (operated by OSR). I'm just a programmer in the trenches trying to understand this stuff.

tedhudek, 2018-06-13:
Hi @fpl9000 sorry but I apparently didn't close out this issue after we updated the related topic.  Maybe because I closed out #451 instead.  Anyway, https://docs.microsoft.com/windows-hardware/drivers/install/kernel-mode-code-signing-policy--windows-vista-and-later- is freshly overhauled, hopefully clearer.  Thanks again for the initial feedback!

closed 2018-06-13 by tedhudek