MicrosoftDocs/windows-driver-docs #448: That's great and all, but what about Symbol downloads for WindowsUpdate.log errors?

tombull89, 2018-04-24:
Hi guys. I'm currently trying to troubleshoot an issue with Windows 10 updates, and I've generated a readable (well) log using the powershell get-windowupdatelog command. I've now got a windowsupdate.log file which is filled with "Unknown GUID—–No Format Information Found" errors. I need to download the Symbol files - but the link at https://msdn.microsoft.com/library/windows/desktop/ee416588.aspx seems to apply to Vision Studio only. I grasp there's something to do with "symchk" as well but this is horribly horribly complicated for something where I only had to open the windowsupdate.log file in Windows 7. Am I missing something here?

---
#### Document details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3fabe759-b55e-91e3-ff40-764310295b5a
* Version Independent ID: 71562214-8b57-b69f-1e02-aff3a7e826e2
* Content: [Download  Windows Symbol Packages for Debugging](https://docs.microsoft.com/en-gb/windows-hardware/drivers/debugger/debugger-download-symbols)
* Content Source: [windows-driver-docs-pr/debugger/debugger-download-symbols.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debugger-download-symbols.md)
* Service: **unspecified**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

duncanmacmichael, 2018-04-24:
Hi @tombull89 , sorry you're having an update issue. Although this topic is aimed at programmers who need symbols for debugging their driver code, I did find this other support topic that might help you: https://support.microsoft.com/en-us/help/3036646/how-to-read-windows-update-logs-in-windows-10-version-1607. If that doesn't solve the problem, then we'd recommend asking a question on the MSDN support forums here: https://social.msdn.microsoft.com/Forums/en-US/home. Sorry, but we're not equipped to handle technical support beyond documentation questions.

From a docs perspective, though, we will check the directions and links on that page to see what we can do to make them clearer. Thanks for bringing this to our attention, and thanks for your interest in improving the docs.

closed 2018-04-24 by duncanmacmichael

tombull89, 2018-04-26:
As we're on the topic of documentation then, there's a link in https://support.microsoft.com/en-us/help/3036646/how-to-read-windows-update-logs-in-windows-10-version-1607 which points to https://docs.microsoft.com/en-us/powershell/module/windows-update/get-windowsupdatelog which is a 404.

EliotSeattle, 2018-04-26:
Working on that broken link. It's another team at MS that handles the support and IT content, but I'll track down the right person. Thanks, @tombull89, for pointing it out.

johnmroberts, 2018-04-27:
Where can I get the symbols for Windows 10 1511 Build 10586? The public ones will not decode the WindowsUpdate.log and all the download links from similar threads a while back are dead.

What was wrong with the old text format? Did you guys look at systemd and think "That's a good idea?"...

Forcing me to upgrading thousands of clients to the latest build of Windows 10 is not a valid answer. ;)

duncanmacmichael, 2018-04-27:
Hi @johnmroberts , sorry you're experiencing a similar issue. Thanks for the question. Because this topic is meant to help Windows driver programmers, we don't have the resources to support IT pros with all the situations where you might need Windows symbols. But, maybe these links can help you.

For any kind of IT-related symbols needs we'd recommend checking out this link: https://support.microsoft.com/en-us/help/3036646/how-to-read-windows-update-logs-in-windows-10-version-1607. For more info about the cmdlet, which is supported starting in Windows 10, see this link: https://docs.microsoft.com/en-us/powershell/module/windowsupdate/get-windowsupdatelog?view=win10-ps. The cmdlet should automatically download the symbol files if you are connected to the Internet.

If that doesn't fix the issue, then we'd recommend asking a question on the MSDN social forums here: https://social.msdn.microsoft.com/Forums/en-US/home?forum=en-US. Also, the blog post linked to in the "Important" box on this page provides more information about the change in the way we distribute public symbols.

Again, thanks for the question. Good luck with your issue.

johnmroberts, 2018-04-30:
Thanks. I'd already tried the information in those links before I made the comment. I've tried clearing out the cache etc, and the symbol files PowerShell is downloading automatically don't work. The WindowUpdate.log that the PowerShell cmdlet creates just contains thousands of lines similar to this :
1601/01/01 00:00:00.0000000 1112  12488                 Unknown( 21): GUID=fa2014d3-4bd2-e53d-471c-3afe586eaa20 (No Format Information found).

I'm hoping the original symbols for Windows 10 1511 would resolve the issue. Unfortunately as the old standalone download has been erased from history by Minitrue I'm a bit screwed!

PBrawn, 2019-06-03:
I'm in the same boat. I had to run a recovery on a Surface 3 tablet (downloaded and ran the recommended 1511 USB installer), and from there, I attempted to upgrade to a supported release (namely 1809) from config manager. Sadly, I get an error that (from what I've read) means there are conflicting updates, but I cannot find out which one is conflicting because I cannot decode the log files. It has taken me 2 days to get to this point and I find it absurd that I can't read the log file that the surface supposedly created. 

I am also getting the reams of incorrect date and unknown GUIDs, but it seems that driver updates appear fine.
So, to summarise:
Fresh install of 1511 (10586.29) from MS downloaded recovery key.
Attempted to install feature update, but this apparently conflicts with another update that neither config manager, nor the surface can tell me the CI_GUID of.
I attempt to look at the windowsupdate log files, but realise they're not right
Trawl through various articles in an attempt to decode the ETL files (learn a bit about symbols in the process)
All recommendations (MS and otherwise) have failed.
I am now about to install a 4Gb 1511 emulator to see if I can convert the ETLs.

This is an obscene amount of work to find out which update is conflicting.
Rather than apologising for the inconvenience, could you please just allow us access to the MSI for this particular release? It's clearly buggy!

PBrawn, 2019-06-03:
p.s. I've not included the wrigmarole of trying to figure out the 0x80240fff error with regards to conflicting updates. MS have nothing on this, yet they deem a 1511 USB recovery release to be adequate for their own product (surface 3)?

duncanmacmichael, 2019-06-05:
@PBrawn Sorry you're having that issue. You should be able to run the [Get-WindowsUpdateLog](https://docs.microsoft.com/en-us/powershell/module/windowsupdate/get-windowsupdatelog) PowerShell command, which will dump the WindowsUpdate.log text file to the desktop. That file is a simple text file that should open in NotePad. For more info about reading that file, see [Windows Update Logs](https://docs.microsoft.com/en-us/windows/deployment/update/windows-update-logs).

duncanmacmichael, 2019-06-05:
Please note that the documentation page on which this thread was created is specifically aimed at downloading symbols for debugging Windows drivers. It is for driver developers and is not aimed at troubleshooting Windows update logs or any other situation that involves logs or symbols.

PBrawn, 2019-06-05:
Apologies Duncan, it wasn't aimed at all you hard working developers! I'm just really frustrated by this issue. The get-windowsupdatelog result is unreadable, despite pointing it to the online symbol repository, or a locally stored one (which is the result of the downloaded one anyway). I think it's simply a case of MS not proividing the correct symbols for this particular OS build.

johnmroberts, 2019-06-05:
> Please note that the documentation page on which this thread was created is specifically aimed at downloading symbols for debugging Windows drivers. It is for driver developers and is not aimed at troubleshooting Windows update logs or any other situation that involves logs or symbols.

Understood.  I realise you are getting the flack for a different team integrating with your code intended for another purpose.

It's a issue for operation IT staff - they can't troubleshoot Windows Update problems any longer on some clients. It might say "all versions of Windows 10", but for machines running older versions of Windows 10 (which were still in extended support when I originally posted April 2018) the symbols download required was no longer available. The event logging got changed from a simple text file to the new method after the download was pulled. Therefore the powershell commands failed unless you had the foresight to stash the symbol files. Which we hadn't not being a driver development shop. Not even archive.org had stashed it.

Out there in the real world we're running LoB systems that might not work on the latest version of Windows 10, so we've got to keep (say) 1511 in use until such time as the supplier update their product. Not everything is agile developed, in the cloud and run by Microsoft/AWS etc.

Luckily no longer a problem for us, we're on 1803 now, but the idiocy caused us some real pain.

TL;DR - My feedback is that Microsoft really need to work on communication so they stop breaking things. I'm going to get back to harassing Microsoft support for breaking most of our websites with the HSTS patch they pushed out three weeks ago that patently hadn't been anywhere near a computer in the UK.

duncanmacmichael, 2019-06-06:
No worries @PBrawn and @johnmroberts. Your situations are understandable. I'm actually not a developer for Windows, though; I'm part of the Windows Driver Kit documentation team, and our team's focus is on providing accurate docs for the intended driver developer audience by working with feature teams to document the various driver technologies and tools. 

But, as you have pointed out, this is a situation where the topic at hand overlaps with other audiences - namely IT pros - who might need Windows symbols from time to time. If the symbols or PowerShell cmdlet aren't working as expected on certain builds, then I suppose the only thing I can do from my end is recommend submitting product feedback using the [Windows Feedback Hub](http://aka.ms/dexex). This particular link automatically sets the product category to Debugging Tools for Windows.

Again, I empathize with your frustrations. The reason I posted that note was more for future readers who might stumble across this thread, just to make it clear that we were the docs team and weren't equipped to troubleshoot the variety of Windows Update log issues that might arise in the wild. We do appreciate you opening the issue, though, as we strive to improve the docs and provide the best information at all times. If we can help future readers who might be able to solve their issue using some of the links in this thread, then it's worth it. :) Thanks again for the feedback.