MicrosoftDocs/windows-driver-docs #454: PowerShell requires quotes around IDs

sdwarwick, 2018-04-26:
would be helpful to remind folks that {guid}  needs to be surrounded by single quotes in powershell

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c2e888de-16af-86ce-3844-b55b519aea27
* Version Independent ID: 97cbb0fb-8c60-afe6-5170-0bfcc75a8613
* Content: [Adding Boot Entries](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/adding-boot-entries#feedback)
* Content Source: [windows-driver-docs-pr/devtest/adding-boot-entries.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/adding-boot-entries.md)
* Service: **unspecified**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2018-04-27:
Thanks @sdwarwick for this idea.  I took a quick glance at the page, but I'm not familiar with this one.  I see a lot of possible places where we could put such a note, but since you ran into the issue yourself, is there a particular spot in there where you think it would be most useful?  You can submit a PR if you like.

tedhudek, 2018-06-13:
Thanks @sdwarwick, I added a parenthetical with this tip to https://docs.microsoft.com/windows-hardware/drivers/devtest/adding-boot-entries.  Should be live in a few minutes.  To see it, just search for PowerShell.

closed 2018-06-13 by tedhudek