MicrosoftDocs/windows-driver-docs #456: WDDM 2.1, 2.2, 2.3 and 2.4 documentation

alessiot89, 2018-04-29:
There is no documentation at all about WDDM 2.1, 2.2, 2.3 and 2.4.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 49631f4a-392a-7b0d-7687-b441d38b1a99
* Version Independent ID: fb471aa4-00c9-a9d6-5314-4b63b1ef4469
* Content: [Windows Display Driver Model (WDDM) Design Guide](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/windows-vista-display-driver-model-design-guide#feedback)
* Content Source: [windows-driver-docs-pr/display/windows-vista-display-driver-model-design-guide.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/windows-vista-display-driver-model-design-guide.md)
* Service: **unspecified**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-04-30:
Thanks, @alessiot89.
@mimisasouvanh, can you look into whether we list the additions in https://docs.microsoft.com/en-us/windows-hardware/drivers/what-s-new-in-driver-development and if we can better surface DDIs by version?

mimisasouvanh, 2018-05-16:
Hi @alessiot89, we are currently working to update the WDDM 2.xx documentation in the coming months. Thanks!

closed 2018-05-16 by mimisasouvanh

vnvsrikanth, 2018-07-26:
When I download the PDF, the bookmarks (navigation pane) is not going to proper chapters in the document. Once you download the PDF, try going to "Introduction to the Windows Display Driver Model(WDDM)" chapter from bookmarks (navigation pane)

404517981, 2021-05-17:
WDDM: Render Only umd dirver is valid for d3d12 ,d3d11,d3d10 with  invoked OpenAdapter10-12(enum to valid adapter through d3d10-12 api ). But d3d9 never do that with kmd dirver  work well.What's different with d3d9 and d3d10-12?