MicrosoftDocs/windows-driver-docs #457: Incorrect sequence for the asynchronous redirection

Initex, 2018-04-30:
According to the https://social.msdn.microsoft.com/Forums/sqlserver/en-US/d7601fe1-9728-4185-bca5-05874a6ffa1f/fwpscompleteclassify-not-working?forum=wfp the sequence for the asynchronous redirection is not correct.
FwpsAcquireWritableLayerDataPointer must be called from the worker function, not from classifyFn!

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4a3e4808-0c6b-8848-6270-8f5e8b810223
* Version Independent ID: d38592f5-3ca4-3d2a-20fd-c8cfd28b58d6
* Content: [Using Bind or Connect Redirection](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/using-bind-or-connect-redirection)
* Content Source: [windows-driver-docs-pr/network/using-bind-or-connect-redirection.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/using-bind-or-connect-redirection.md)
* Service: **unspecified**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

duncanmacmichael, 2018-04-30:
Hi @SerusDev , thanks for the feedback. I'll investigate the answer to this and get back to you shortly.

duncanmacmichael, 2018-05-02:
Hi again @SerusDev , after confirming the answer I've updated the page to have the correct order for the steps. The changes should be live shortly. Thanks so much for your feedback in improving the docs!

closed 2018-05-02 by duncanmacmichael