MicrosoftDocs/windows-driver-docs #458: deleting profiles

jszabo98, 2018-04-30:
Deleting a user profile will break this system.  Either attempts to add a printer will fail, or unwanted printers will keep coming back.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2f5319b8-387d-0944-6ae7-6d71e827a878
* Version Independent ID: 0b27703e-e073-ecd4-424d-6c09a792f1d3
* Content: [Known Issues with Client-Side Rendering](https://docs.microsoft.com/en-us/windows-hardware/drivers/print/known-issues-with-client-side-rendering)
* Content Source: [windows-driver-docs-pr/print/known-issues-with-client-side-rendering.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/print/known-issues-with-client-side-rendering.md)
* Service: **unspecified**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-04-30:
Also, see update so this is more than Windows Vista. =:-)

TimvdLooy, 2018-09-28:
This option in combination with secure print can cause the Windows 10 print app to not work correctly.

barrygolden, 2019-08-07:
This topic has been updated. 

Thanks @jszabo98 and @TimvdLooy for your comments on this topic.

closed 2019-08-07 by barrygolden