MicrosoftDocs/windows-driver-docs #464: Does Universal Driver ( DCHU design principles)  requirement for HLK?

armstrong-charles, 2018-05-01:
I have a question regarding universal driver, design principles are very good, it make sense, it would truly make a driver as a universal driver that can run on any UWP supported platform, as well as device manufacturer and and system builder can seamlessly work together to extend the device funcitonalities.

What is the road map,  will it become a requirement for HLK, does it apply to specific type of class drivers? e.g. do you think that at some point, printer driver will be universal driver?
Please provide some insights.
Thanks

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2c5fa76b-6c2e-8f5e-7e96-ceb1c726842b
* Version Independent ID: 48a29c52-9907-9b1a-1dc2-3365c2d021d0
* Content: [Getting Started with Universal Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/getting-started-with-universal-drivers)
* Content Source: [windows-driver-docs-pr/develop/getting-started-with-universal-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/develop/getting-started-with-universal-drivers.md)
* Service: **unspecified**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2018-06-13:
Sorry for the delay on this one, @armstrong-charles.  Assigning this one to @barrygolden, who should be able to tell us if print drivers can be universal.

armstrong-charles, 2018-10-09:
Any update on this?
Thanks

Dark4Codrutz, 2019-01-21:
@barrygolden, I would like to know the answers on these topics as well as to further understand the possibilities of universal drivers.

barrygolden, 2020-11-04:
Print drivers are not Universal drivers. To see examples of types of drivers that are Universal drivers, you can look at the driver samples on GitHub:

https://github.com/microsoft/Windows-driver-samples

If a driver sample is Universal, it will contain several `<DriverTargetPlatform>Universal</DriverTargetPlatform>` entries in the sample's .vcxproj file.

closed 2020-11-04 by barrygolden