MicrosoftDocs/windows-driver-docs #469: So whats the point of this

actionaer, 2018-05-03:
asking

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 17e495a6-0bb0-5f8c-fd6a-948f9ecec253
* Version Independent ID: 56480873-4b16-52c6-9404-4c09bdfdf8c9
* Content: [Synchronization Techniques](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/synchronization-techniques)
* Content Source: [https://cpubwin.visualstudio.com/_git/drivers?path=%2fwindows-driver-docs-pr%2fkernel%2fsynchronization-techniques.md&version=GBlive&_a=contents](https://cpubwin.visualstudio.com/_git/drivers?path=%2fwindows-driver-docs-pr%2fkernel%2fsynchronization-techniques.md&version=GBlive&_a=contents)
* Service: **unspecified**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-05-03:
Hi @actionaer.
I'm not quite clear about what you are asking. Are you looking for more info about why synchronization techniques are important, what this particular page is setting out to do, or is it something else?
Thanks, Eliot

EliotSeattle, 2018-07-18:
Closing with no response.

closed 2018-07-18 by EliotSeattle