MicrosoftDocs/windows-driver-docs #471: Possible cause of this.

kwilliams1987, 2018-05-04:
This bug check is caused when you change the Software Guard Extensions (SGX) mode of the UEFI BIOS from Software Controlled to Enabled.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 70fe0f26-49e2-df0d-1864-f325aa4c0cb6
* Version Independent ID: b99d000e-7805-5cdd-768f-11249ad48fb8
* Content: [Bug Check 0x65 MEMORY1_INITIALIZATION_FAILED](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x65--memory1-initialization-failed)
* Content Source: [https://cpubwin.visualstudio.com/_git/drivers?path=%2fwindows-driver-docs-pr%2fdebugger%2fbug-check-0x65--memory1-initialization-failed.md&version=GBlive&_a=contents](https://cpubwin.visualstudio.com/_git/drivers?path=%2fwindows-driver-docs-pr%2fdebugger%2fbug-check-0x65--memory1-initialization-failed.md&version=GBlive&_a=contents)
* Service: **unspecified**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2018-06-25:
Planning to consolidate the debugging issues and get with the dev team this week.

DOMARS, 2018-06-28:
@kwilliams1987  – Thank you for opening this issue. We use input such as this, along with Windows telemetry and other data, to track trends in stop code behaviors. We add information to the topic if we find the behavior is consistent over time, so we’ve noted your findings and will look into the specific behavior.

closed 2018-06-28 by DOMARS