MicrosoftDocs/windows-driver-docs #494: Submissions will enforce the EV-only requirement later this year

OzBob, 2018-05-15:
This is not clear whether 'attestation driver signing' submissions will also require EV-only.
There is also very little information about how to obtain an EV certificate or even what a 'standard' certificate is used for.

---
#### Document details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 93d3f097-d81b-66c2-4835-d63758687d34
* Version Independent ID: fe9d1cdd-966e-89b5-c66b-60531b0ec655
* Content: [Get a code signing certificate](https://docs.microsoft.com/en-gb/windows-hardware/drivers/dashboard/get-a-code-signing-certificate#feedback)
* Content Source: [windows-driver-docs-pr/dashboard/get-a-code-signing-certificate.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/get-a-code-signing-certificate.md)
* Service: **windows-hardware**
* GitHub Login: @aahill
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-05-15:
Thanks, @OzBob 
@aahill  can you investigate? Thx!

aahill, 2018-05-25:
Hi @OzBob , sorry for the delay in responding to you. Thanks for reaching out, and sorry this topic has caused you confusion!

The [attestation signing](https://docs.microsoft.com/windows-hardware/drivers/dashboard/attestation-signing-a-kernel-driver-for-public-release) topic mentions that an EV certificate is required for attestation signing. I've added this to the code signing table and should be live in 24 hours. 

Regarding your question about certificates, would you let me know what sort of information you'd like us to include? On the topic you mention, we have some of the differences between standard and EV  certificates below the requirements table, along with a list of certificate authorities. If you're looking more for text about general code signing, I can certainly include that in the topic.

Thanks!
-Aaron

OzBob, 2018-05-28:
Thanks for responding @aahill

Following on from two recent dotnetrocks podcast giving a bad review on the ease for developers to sign their assemblies for distribution. (https://www.dotnetrocks.com/?show=1547 and https://www.dotnetrocks.com/?show=1535)

I began searching for whether I needed a EV or Standard certificate to put code through the microsoft store and also for distrbution through MSIXs, or whether I could use my domain's private SSL certificate to sign my assemblies. 

Although there is some information in the docs, it hides the deadend which is :
http://aka.ms/rootcert says: "A CA must fill out the [PDF form] application [on behalf of the applicant] located at http://aka.ms/rootcertapply and email ... to msroot@microsoft.com."
A physical old school non-interactive PDF.

I'm not sure what I can do with a Standard certificate if I need an EV cert to join the program/dashboard, or whether (for my purposes) I need to join the Hardware Dev Center.

I do appreciate your assistance, and as you can imply from the DotNetRocks sentiment, the Attestation Signing process for application developers who are not hardware driver distributors, is not clear.

aahill, 2018-05-30:
Thanks for the podcast links @OzBob - I'll be sure to check them out and pass what I learn to our engineering team. I agree that its a bit confusing; I'm still looking into the process listed on aka.ms/rootcert; as far as I know you shouldn't have to do anything with a pdf - I'm investigating this.

If I'm understanding right, you're looking more for info about distributing app packages, rather than driver packages, correct? Not a problem; I have limited experience in that area but from what I understand, the Microsoft store will sign the app package for you with a trusted certificate when you publish it. Here's the [documentation](https://docs.microsoft.com/windows/uwp/publish/) relating specifically for app publication.

If you're interested in manually signing your package with your domain's certificate (usually for testing purposes, or distributing the app outside the store) , this topic may help:
https://docs.microsoft.com/en-us/windows/uwp/packaging/sign-app-package-using-signtool

I'll continue looking into this; let me know if these resources help you out!

closed 2018-06-05 by aahill