MicrosoftDocs/windows-driver-docs #496: What will happen if MO ignores SOSA?

hli88, 2018-05-15:
What will happen if MO ignores SOSA? For example, MO provides their users APN names and let user manually import them into Win10 cellular settings. During my test, user still can access to the network. Will SOSA influence switching different APNs (multiple PDP contexts) when an application wants to use a different APN for data transfer?  

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 73249166-42e2-1205-a842-818eb35eccad
* Version Independent ID: e50887fc-f2b1-9e46-7747-c32f52de952f
* Content: [COSA overview](https://docs.microsoft.com/en-us/windows-hardware/drivers/mobilebroadband/cosa-overview)
* Content Source: [windows-driver-docs-pr/mobilebroadband/cosa-overview.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/mobilebroadband/cosa-overview.md)
* Service: **windows-hardware**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **windowsdriverdev**

duncanmacmichael, 2018-05-15:
Hi @hli88 , good question. MOs use COSA to provision devices. For example, if AT&T works with us to set up COSA for their network, then when a customer inserts an AT&T SIM card and adds the device to their account then it will work with AT&T's network in the way that AT&T specified. COSA is a database in the format of a provisioning package file that tells Windows what settings to use when talking to the MO's network. It is broad-sweeping and includes things like MO carrier branding, mobile hotspot settings, etc.

As far as how COSA influences different APNs, I don't believe it has a bearing on that but I'll check for you.

duncanmacmichael, 2018-05-15:
Hi @hli88 , I got some confirmation for my answer. COSA is indeed provided by MOs normally MOs are the ones who dictate how device settings should be configured while operating on their network, including the APN to which the device is connecting.

However, if a user such as yourself wants to overwrite the MO settings, you can manually add an APN as described here: https://support.microsoft.com/en-us/help/10739/windows-10-cellular-settings

There are also ways to overwrite the APN in an Enterprise setting but I don't think that applies to you.

Anyway, hope this helps. You setting an APN provided by the MO should work fine regardless of COSA. Thanks for the question.

closed 2018-05-15 by duncanmacmichael