MicrosoftDocs/windows-driver-docs #504: Waiting to reconnect..

lawabider, 2018-05-16:
I've tried everything but can't get it to work.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9564a660-e978-17e0-b1ab-885eb9345046
* Version Independent ID: 97c67bc0-5e7d-01ff-f4ba-1c626820614c
* Content: [Setting Up Network Debugging of a Virtual Machine Host](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/setting-up-network-debugging-of-a-virtual-machine-host)
* Content Source: [windows-driver-docs-pr/debugger/setting-up-network-debugging-of-a-virtual-machine-host.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/setting-up-network-debugging-of-a-virtual-machine-host.md)
* Service: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

EliotSeattle, 2018-05-16:
Closed Issue 503 as a dupe of this one. Comment from 503 was:
"I can't get it to connect...sigh. I've tried restarting the target, opening firewalls."

DOMARS, 2018-06-18:
I don't see a question about the documentation here, so I am closing this Issue as not actionable. If you have a question about using the debugger, your best bet is to look for help in Stack Overflow, the [Windows Hardware WDK and driver development forum](https://social.msdn.microsoft.com/Forums/windowshardware/en-US/home?forum=wdk), or the [OSR NTDEV forum](http://www.osronline.com/cf.cfm?PageURL=showlists.cfm?list=NTDEV). This feedback mechanism is for requests about the documentation. 

We are working on a new KDNET VM debugging topic that may be helpful. Hope to have that published in a week or so. It will be called "Setting Up Network Debugging of a Virtual Machine - KDNET".

closed 2018-06-18 by DOMARS