MicrosoftDocs/windows-driver-docs #505: Is this still available?

applesauce49, 2018-05-16:
I'm using Windows Driver Kit version 10.0.17110.1002 and these options are not available in the Driver->Test Suite.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 41f5a94c-f793-46dc-4045-e90d30777c25
* Version Independent ID: da4260b3-5e23-4e76-ca87-18acf6f288e9
* Content: [How to Perform Fuzz Tests with IoSpy and IoAttack](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/how-to-perform-fuzz-tests-with-iospy-and-ioattack)
* Content Source: [windows-driver-docs-pr/devtest/how-to-perform-fuzz-tests-with-iospy-and-ioattack.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/how-to-perform-fuzz-tests-with-iospy-and-ioattack.md)
* Service: **unspecified**
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

DOMARS, 2018-05-17:
@applesauce49 - Can you also let us know the version of the HLK that you have installed? As you may know, the HLK contains test content, so knowing the HLK version will help us better focus our investigation.

applesauce49, 2018-05-17:
I was not aware of that.  I just installed downloaded and installed HLK 1803 (10.1.17134.1)

DOMARS, 2018-05-18:
@applesauce49 - Thanks for that information, we are investigating.

DOMARS, 2018-06-19:
I investigated with the product team and found that IOSpy and IOAttack are not currently shipping. It is possible that they may return in a future version of the kits, but the plans are not certain at this point. 

I will work to add this information to the docs pages, so thanks so much for pointing out this issue.

As an alternative, you may want to investigate the fuzz tests in the HLK. For example the DF - Fuzz random IOCTL test described here:
https://docs.microsoft.com/en-us/windows-hardware/test/hlk/testref/236b8ad5-0ba1-4075-80a6-ae9dafb71c94

Thanks again for pointing out this issue, I will work with product team to make updates to the related topics.

closed 2018-06-19 by DOMARS