MicrosoftDocs/windows-driver-docs #509: Dpinst install

Michael-D-Sun, 2018-05-18:
Hi MSFT team,
After I add BuildNumber decoration, I notice that the driver can't be installed by Dpinst which I get from WDK 8.1.  I tried to find Dpinst in the new WDK, but I didn't find it. The Dpinst doesn't support this inf directive. Is it the right behavior?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9445c977-9f8c-00c6-68fa-b575ab325fcf
* Version Independent ID: 58963a0b-6f4b-b6ee-f454-7e72e68d7bef
* Content: [INF Manufacturer Section](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/inf-manufacturer-section)
* Content Source: [windows-driver-docs-pr/install/inf-manufacturer-section.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/inf-manufacturer-section.md)
* Service: **unspecified**
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

Michael-D-Sun, 2018-05-28:
Anybody can help on this item? In the inf manufacture section the target os is 10.0...14393. Dpinst can't find the right section to install the driver. Is it the right behavior? And is there any replacement?

tedhudek, 2018-05-29:
Hi @Michael-D-Sun this is more a product question than a doc issue. I recommend trying [Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
or [ntdev forum](http://www.osronline.com/showlists.cfm?list=ntdev)

Best of luck with this issue!

closed 2018-05-29 by tedhudek