MicrosoftDocs/windows-driver-docs #510: Wandering in the Hyperlink desert.

mfstanton, 2018-05-18:
The sections begins with "In this section..." and all I get is a bunch of hyperlinks that point to another page. Do you realize how painful it makes this documentation to follow? I should be able to follow it like a book by simply scrolling down through the pages. Someone went nuts with hyperlinks because they didn't know how to write a document.   Following a hyperlink to a page that is simply a list of hyperlinks is pointless and from a professional viewpoint quite disrespectful of my time as a reader.
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 8c516129-2576-8860-173c-2157273e24a5
* Version Independent ID: 5d4dafdf-0f66-6f77-55b5-05387a7cb8b5
* Content: [Concepts for all driver developers](https://docs.microsoft.com/en-us/windows-hardware/drivers/gettingstarted/concepts-and-knowledge-for-all-driver-developers)
* Content Source: [windows-driver-docs-pr/gettingstarted/concepts-and-knowledge-for-all-driver-developers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/gettingstarted/concepts-and-knowledge-for-all-driver-developers.md)
* Service: **unspecified**
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-05-18:
Thanks, @mfstanton. Sorry that this page is causing you aggravation. 
@PageWriter-MSFT  and I will work on redesigning how this content fits with the larger story of getting started. If you have specific suggestions, we'd love to hear them. Thanks! Eliot

mfstanton, 2018-05-19:
Thank you Sir. I will keep in touch. As someone who has written user documentation from an Engineer’s viewpoint when I was at Unisys, I understand the problem with getting the right amount of content and detail into the opening and subsequent sections. The biggest problem I have is the excessive use of hyperlinked text that doesn’t get me to more than a paragraph or two of information or simply an ordered list of more hyperlinks. That can be really annoying after the third or fourth iteration without actually getting to the meat of a technical document.

dxshindeo, 2018-12-11:
Logged in to comment - I'm a newcomer into windows programming world (need to learn about driver programming), and I completely disagree with you @mfstanton.

When I read a guide, I expect the information the be condensed as much as possible, because that way I can absorb more of it without feeling overwhelmed. Of course, if I want to have more info, that's where hyperlinks come in. If it would be your way, when I open a page, I would be greeted by a 5 minute long scrolling sheet of content explaining everything in crucial detail. Nobody wants that. That would only hurt the learning process.

I strongly believe that the way things are documented right now is the right way to go. I've been reading here for over an hour, knowledge is being spoon fed to me. 

ggwp authors ;)

EliotSeattle, 2018-12-12:
The page is now a TOC entry only. It unfolds to show the topics in the node, but does not have content in it (e.g., nothing but links content), unless you go there via direct link or bookmark.

Should be live this afternoon.

closed 2018-12-12 by EliotSeattle