MicrosoftDocs/windows-driver-docs #514: More possible values for Parameter 1

thierryfranzetti, 2018-05-23:
In latest versions of Windows 10 kernel, Parameter 1 can have many more values than the ones currently documented on this page. Adding some documentation for these new values would help diagnosing this system crash.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5e82f821-f508-03c3-2435-06217f01a61f
* Version Independent ID: 96f99bcc-9de6-ec2d-de9e-e162d5073bf6
* Content: [Bug Check 0x139 KERNEL_SECURITY_CHECK_FAILURE](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check---bug-check-0x139-kernel-security-check-failure)
* Content Source: [windows-driver-docs-pr/debugger/bug-check---bug-check-0x139-kernel-security-check-failure.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check---bug-check-0x139-kernel-security-check-failure.md)
* Service: **unspecified**
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2018-06-26:
@thierryfranzetti - Thanks for this request. I'm checking with the dev team to see if we can list these additional values for parameter 1.

DOMARS, 2018-06-27:
@thierryfranzetti - I was able to work with the dev team to add additional values for parameter 1. Thanks for the request.

closed 2018-06-27 by DOMARS