MicrosoftDocs/windows-driver-docs #515: Altitude of fileinfo

lawlxy, 2018-05-23:
The altitude of fileinfo is 45000 in this page. The altitude of FileInfo in my computer (RS4 Win10 Enterprise, fltmc) is 40500. Which one is correct? 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0a2d558e-1073-85d5-f863-5bdbf6e3ddc6
* Version Independent ID: 62e4d735-803f-7608-154b-7032dc1ceb3d
* Content: [Allocated Altitudes](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/allocated-altitudes#span-id180000-189999fsfilterhsmspanspan-id180000-189999fsfilterhsmspanspan-id180000-189999fsfilterhsmspan180000---189999-fsfilter-hsm)
* Content Source: [windows-driver-docs-pr/ifs/allocated-altitudes.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/ifs/allocated-altitudes.md)
* Service: **unspecified**
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-05-23:
Thanks, @lawlxy 
The writer for this area is on vacation through the middle of next week. If you need the info immediately, I suggest posting in the [Windows Hardware WDK and Driver Development forum](https://go.microsoft.com/fwlink/p/?LinkId=613308)

aahill, 2018-07-19:
Hi @lawlxy, sorry for the delay. I checked and 40500 is the correct altitude; this seems to be a typo. I'll update the documentation. Thanks!

closed 2018-07-19 by aahill