MicrosoftDocs/windows-driver-docs #517: White space

mfstanton, 2018-05-23:
Much too much whitespace and noise on the page. The sidebars take up too much real Estate and stretching the page only makes the margin whitespace expand without expanding the content.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: dd5bbdd7-26e8-d090-e5cb-39a08df73689
* Version Independent ID: 5e7eebfc-3a1c-8979-02df-42a0cbcbc177
* Content: [IRP\_MJ\_CREATE](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/irp-mj-create)
* Content Source: [windows-driver-docs-pr/ifs/irp-mj-create.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/ifs/irp-mj-create.md)
* Service: **unspecified**
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-05-23:
Hi @mfstanton 
We are using a templated design that's employed by much of the Microsoft developer content--millions of pages total. I can pass your feedback on to the designers, but first, just to make sure, are you seeing the same layout I see in the attached screenshot?
[image]

mfstanton, 2018-06-08:
Yep, that is what I am seeing. I realise that Pubs people have mandated standards but they are not the ones who have to get detail out of a page or go through a lot of pages to get to meaningful content. . Often there is not enough content on a page to make it worth navigating to. The only saving grace is that some of the pages have an associated PDF that can be downloaded. Even those have a lot of white space.

mfstanton, 2018-06-08:
Think about who your audience is. We don't have time to riff through a lot of pages to find what we want.

EliotSeattle, 2018-06-27:
Thanks, @mfstanton 
I've spoken with the publishing team and have looked over the content. I believe that you and I are talking about different definitions of whitespace. While the layout of a page (column width, TOC size, header, etc.) are mandated to keep the site consistent, what I think you're really talking about is the amount of content on each page. If I hear you correctly, pages like [Reference](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/ifs-reference) are irksome, pages like [Miscellaneous Information](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/miscellaneous-information) are a waste of your time, and pages like [Locating Disk Images Correctly](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/locating-disk-images-correctly) should be incorporated into another topic to make larger building blocks of content.

If I understand you correctly, I get it. These are artifacts of our XML-based authoring tools that we just migrated out of and into markdown and GitHub. We are going through the various content sections and making things more usable, but we had to migrate the structure at first so as not to lose content. It will likely take some time, but we have work items to attack these issues.

Let me know if we are on the same page here or help me to understand where your frustrations lie in this.
Thanks,
Eliot

mfstanton, 2018-06-28:
Yes, Eliot,  that is pretty close to what I was thinking. There are parts of the miscellaneous information that are helpful but that seems to be a repository for a lot of fluff too.  Having to jump to a separate page just to see it is a waste of clicks and bandwidth. Yes the pages like “Reference” that have little or no content should be gone.  I am asking for more meat and less fluff. You have captured my frustration quite well. That kind of philosophy for works fine for a place where you are trying to get as much advertising content (3 lines of news article text surrounded by 15 adverts) in front of a user. It doesn’t help a developer get their job done.   I guess it depends on what the objective is for the web page.
 
I really like your evaluation of what I was trying to say.