MicrosoftDocs/windows-driver-docs #534: Updating Device Firmware using Windows Update: sample code and questions.   If  SSD WU is recommended by microsoft?

cai6jing, 2018-05-30:
[Enter feedback here]
1.   BUG 17170213
If  SSD firmware WU is recommended by microsoft?

2. The last 3rd bullet: 

 In the diagram immediately above, the firmware driver can look for the parent device to find the primary device with which it needs to communicate

1. How to look for the parent device to find the primary device with which it needs to communicate and how to communicate?  ( sample code?)

 2. How to get/scan disks’ instance? How to send command to them?  (can use UMDF2 ? or need to useKMDF? )

 3.   After finish install firmware update driver, there are 2 devices ( please check the attached pic): is it expected?  
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 07d25e12-2213-9c84-2fe9-ee2f32e82a27
* Version Independent ID: 6ff8eec2-25d4-f1af-a155-74cfa99ad7df
* Content: [Updating Device Firmware using Windows Update](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/updating-device-firmware-using-windows-update)
* Content Source: [windows-driver-docs-pr/install/updating-device-firmware-using-windows-update.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/updating-device-firmware-using-windows-update.md)
* Service: **unspecified**
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2018-06-14:
This was handled separately.

closed 2018-06-14 by tedhudek

cai6jing, 2018-06-14:
Sorry， I do not understand your comment: could you please elaborate? any direct answer?

Thanks!

NabilFates, 2018-07-26:
cai6jing,

A bug was filed on Microsoft to address this privately. Please refer to that bug for further communications on this issue.

Thanks,
Nabil