MicrosoftDocs/windows-driver-docs #536: Any API to distinguish base driver and extension driver/component driver

guliteb, 2018-06-01:
I am developing an application to manage driver update. I would like to know if there is any API to get extension driver information for a specific device/Hardware ID. I need to check the driver version of the installed extension driver and determine if a new driver is applicable. I go through SetupAPI and don't find a suitable API.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1cc2449f-18a8-dfb2-c372-d140b46239a9
* Version Independent ID: 19f4054d-bc9d-a269-0274-9316d5936176
* Content: [Working with Extension INFs in the Windows Hardware Dashboard](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/submit-dashboard-extension-inf-files)
* Content Source: [windows-driver-docs-pr/dashboard/submit-dashboard-extension-inf-files.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/submit-dashboard-extension-inf-files.md)
* Service: **unspecified**
* Product: **windows-hardware**
* GitHub Login: @aahill

NabilFates, 2018-06-01:
Try querying DEVPKEY_Device_ExtendedConfigurationIds using SetupAPI (or Devices.Enumartion or configmgr), and you should be able to get that information.

guliteb, 2018-06-04:
Hi Nabil,
Thanks for your reply.

I cannot find any document on Microsoft Docs site about property DEVPKEY_Device_ExtendedConfigurationIds. It is only defined in devpkey.h. Can you guide me to get more explanation on this property?

To give a try, I used the following code to get the buffer required for DEVPKEY_Device_ExtendedConfigurationIds. But when it returns FALSE, it doesn't return ERROR_INSUFFICIENT_BUFFER as expected. Instead, GetLastError() always return ERROR_NOT_FOUND (1168).

However, when I switch to DEVPKEY_Device_HardwareIds, which is also DEVPROP_TYPE_STRING_LIST type, the same code can get me the buffer size and then I can call SetupDiGetDeviceProperty again to get the HWIDs.

Is this Extended property valid for all devices? What kind of information is stored in this property?

 // Get the required buffer size
 bResult = SetupDiGetDeviceProperty(​
  DeviceInfoSet,​
  &DeviceInfoData,​
  &DEVPKEY_Device_ExtendedConfigurationIds,​
  //&DEVPKEY_Device_HardwareIds,​
  &PropType,​
  NULL,​
  0,​
  &Size,​
  0);

Best Regards
Liang Junming

guliteb, 2018-06-04:
Actually, I want to check the version of a base INF/driver and an extension INF/driver, as one device may have both base and extension (even component driver) with different version number. I need to get all versions of those drivers on one device and then determine which should be updated.

Thanks.

Best Regards
Liang Junming
________________________________
发件人: Liang Junming <guliteb@hotmail.com>
发送时间: 2018年6月4日 7:22
收件人: MicrosoftDocs/windows-driver-docs; MicrosoftDocs/windows-driver-docs
抄送: Author
主题: 答复: [MicrosoftDocs/windows-driver-docs] Any API to distinguish base driver and extension driver/component driver (#536)

Hi Nabil,
Thanks for your reply.

I cannot find any document on Microsoft Docs site about property DEVPKEY_Device_ExtendedConfigurationIds. It is only defined in devpkey.h. Can you guide me to get more explanation on this property?

To give a try, I used the following code to get the buffer required for DEVPKEY_Device_ExtendedConfigurationIds. But when it returns FALSE, it doesn't return ERROR_INSUFFICIENT_BUFFER as expected. Instead, GetLastError() always return ERROR_NOT_FOUND (1168).

However, when I switch to DEVPKEY_Device_HardwareIds, which is also DEVPROP_TYPE_STRING_LIST type, the same code can get me the buffer size and then I can call SetupDiGetDeviceProperty again to get the HWIDs.

Is this Extended property valid for all devices? What kind of information is stored in this property?

 // Get the required buffer size
 bResult = SetupDiGetDeviceProperty(​
  DeviceInfoSet,​
  &DeviceInfoData,​
  &DEVPKEY_Device_ExtendedConfigurationIds,​
  //&DEVPKEY_Device_HardwareIds,​
  &PropType,​
  NULL,​
  0,​
  &Size,​
  0);

Best Regards
Liang Junming

NabilFates, 2018-06-04:
Hey Liang,

Unfortunately, I think I spoke too soon. The DEVPKEY I mentioned is indeed in the public header, but its use is mostly internal. With the available APIs and properties in the latest version of the OS, I don't believe it is possible to achieve this scenario. My suggestion is to instead use DiInstallDriver with the extension INF you're trying to install. This API will take care of version checking for you, installing the Extension INF on any applicable devices that don't have that driver (or a higher version) installed. 

As a side note, I'd like to reinforce that Microsoft would like to get as much users as possible leveraging Windows Update (and our entreprise management tools, such as SCCM) for automatic driver acquisition.

Thanks,
Nabil

guliteb, 2018-06-06:
Hi Nabil,
I appreciate your detail reply and explanation. It is very helpful.

Anyway, I would hope there are more APIs for developers to work with Universal INF/Driver in the future as it is going to be widely used on Windows 10.

Thanks.

closed 2018-06-06 by tedhudek