MicrosoftDocs/windows-driver-docs #538: portcls.sys incorrectly processes buffer size query requests

emuzychenko, 2018-06-02:
In some cases, PortCls port drivers return STATUS_BUFFER_TOO_SMALL instead of STATUS_BUFFER_OVERFLOW in response to a property request with zero output buffer address and size (a way to determine required output buffer size). Required buffer size is not returned in such cases.

For example, it happens with KSPROPERTY_PIN_CONSTRAINEDDATARANGES in all three PortCls port drivers (WaveRT, WaveCyclic, WavePci). Miniport property handler is specified in miniport automation table but never called. Miniport automation table is actually processed by PortCls (if property/handler descriptor is removed from the table, PortCls returns STATUS_NOT_FOUND for the same request).

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 91bb7a22-c1ce-29ee-28c2-5711c047a792
* Version Independent ID: b3e69a56-c02b-7b85-f14f-e888f86c9784
* Content: [Audio Property Handlers](https://docs.microsoft.com/en-us/windows-hardware/drivers/audio/audio-property-handlers)
* Content Source: [windows-driver-docs-pr/audio/audio-property-handlers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/audio/audio-property-handlers.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **windowsdriverdev**

DOMARS, 2018-06-29:
@emuzychenko  - I’m checking with the audio dev team to see if they can comment on what you are seeing. Thanks for passing on your observations, and your interest in improving these pages.

DOMARS, 2018-07-30:
@emuzychenko - I have heard back from the dev team and I'm working on updates to the doc.

DOMARS, 2018-08-08:
@emuzychenko - There was additional discussion across a couple of dev teams, and we have settled on updated text which is now live. Additional links were also added that further discuss the general situation. Thanks again for passing on your observations, and your interest in improving these pages.

EliotSeattle, 2018-08-09:
@DOMARS

DOMARS, 2019-06-05:
@emuzychenko - I have not been able to gain clarity on the buffer situation that you describe in your code with the dev team.  As another resource you might present your issue to the OSR driver forum to see if anyone else has run into the issue with buffering in audio drivers that you describe. 

[OSR's ntdev List](https://community.osr.com/categories/ntdev)

Good luck in your development efforts and I hope that you were able to find a way to work around the issues that you encountered.

closed 2019-06-05 by DOMARS