MicrosoftDocs/windows-driver-docs #550: What completion params to use with EvtIoRead and EvtIoWrite completions?

andyraf, 2018-06-05:
I looked through MSDN, didn't find anything (at least, not easily). Could you publish exactly how to return the #bytes read or written for each of these APIs?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f2b6f03e-25a2-08a7-5f55-5b1eaef879d8
* Version Independent ID: 1d941f6c-10c0-e4ed-99b2-ea23332c6102
* Content: [Completing I/O Requests](https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/completing-i-o-requests#feedback)
* Content Source: [windows-driver-docs-pr/wdf/completing-i-o-requests.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/wdf/completing-i-o-requests.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2018-06-13:
@Shyamal-V @wm1 any thoughts?

wm1, 2018-06-13:
Use [WdfRequestCompleteWithInformation(Request, status, bytesCopied);](https://github.com/Microsoft/Windows-driver-samples/blob/df271b80bdbb556707d9b4af1b06151ded561884/serial/VirtualSerial2/queue.c#L542)

tedhudek, 2018-06-14:
Thanks @wm1!  Closing this out, @andyraf, let us know if we can help otherwise.

closed 2018-06-14 by tedhudek