MicrosoftDocs/windows-driver-docs #552: Not working

LarryK348, 2018-06-06:
after following these instructions on 16299.rx3_release.170928-1534 windbg reports:
The system does not support local darnel debugging.

Local kernel debugging requires Administrative
privileges, and is not supported by WOW64.
Only a single local kernel debugging session can run at a
time.
Local kernel debugging is disabled by default. You must run
bcdedit -debug on and reboot to enable it.

bcdedit was update successfully and system rebooted, windbg was started with at Administrative level.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 97eed641-8cd3-4359-aee7-2f14acd28b24
* Version Independent ID: c0c6d81e-4487-cd62-9dcd-9ce3b14f98e0
* Content: [Setting Up Local Kernel Debugging of a Single Computer Manually](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/setting-up-local-kernel-debugging-of-a-single-computer-manually)
* Content Source: [windows-driver-docs-pr/debugger/setting-up-local-kernel-debugging-of-a-single-computer-manually.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/setting-up-local-kernel-debugging-of-a-single-computer-manually.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

EliotSeattle, 2018-06-06:
Thanks, @LarryK348 
@DOMARS can you please look into whether this repros?

DOMARS, 2018-06-19:
Local kernel debugging was successful for me and continues to be functional. Make sure that you are using the correct "bitness" of the debugger to match the version of Windows. If you are running 64-bit Windows, use the debugger in the x64 directory. 
For further help, your best bet is to look for help in Stack Overflow, the [Windows Hardware WDK and driver development forum](https://social.msdn.microsoft.com/Forums/windowshardware/en-US/home?forum=wdk), or the [OSR NTDEV forum](http://www.osronline.com/cf.cfm?PageURL=showlists.cfm?list=NTDEV).

closed 2018-06-19 by DOMARS