MicrosoftDocs/windows-driver-docs #557: What "register" is mentioned?

emuzychenko, 2018-06-08:
"...the client loads the structure with values that indicate the preferred base address for the register"

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: cab7c4b0-b812-7d3c-9318-a238510bb27a
* Version Independent ID: b7c5599f-7cc5-2b20-848d-0c494a369171
* Content: [KSPROPERTY\_RTAUDIO\_GETREADPACKET](https://docs.microsoft.com/en-us/windows-hardware/drivers/audio/ksproperty-rtaudio-getreadpacket)
* Content Source: [windows-driver-docs-pr/audio/ksproperty-rtaudio-getreadpacket.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/audio/ksproperty-rtaudio-getreadpacket.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-06-08:
@DOMARS same as #558

DOMARS, 2018-06-25:
@emuzychenko - Thanks for catching this. The summary text was incorrect and was missed in review. The new summary reads:

*Before sending the request, the client loads the structure with values that indicate the packet number, packet length and other information.*

Thanks again for seeing this error and help making our docs better.

closed 2018-06-25 by DOMARS