MicrosoftDocs/windows-driver-docs #558: What "register" is mentioned?

emuzychenko, 2018-06-08:
"...the client loads the structure with values that indicate the preferred base address for the register"

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1575acfe-5f19-453b-6e85-ae1fc70920c0
* Version Independent ID: d6d8372a-2759-ddab-6932-1b97fa82c633
* Content: [KSPROPERTY\_RTAUDIO\_SETWRITEPACKET](https://docs.microsoft.com/en-us/windows-hardware/drivers/audio/ksproperty-rtaudio-setwritepacket)
* Content Source: [windows-driver-docs-pr/audio/ksproperty-rtaudio-setwritepacket.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/audio/ksproperty-rtaudio-setwritepacket.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-06-08:
Thanks, @emuzychenko for pointing this out.
@DOMARS can you define the register or link to the topic that does?

emuzychenko, 2018-06-09:
Additionally, first link to KSRTAUDIO_SETWRITEPACKET_INFO points to KSRTAUDIO_HWREGISTER description. There are several RTAudio-related pages with such bug. Definitely, the person who created these pages, initially copied them from KSRTAUDIO_HWREGISTER page.

DOMARS, 2018-06-26:
@emuzychenko - Thanks for catching this. The summary text was incorrect and was missed in review. The new summary reads:

*Before sending the request, the client loads the structure with values that include the packet number, packet length and other information.*

The link to KSRTAUDIO_SETWRITEPACKET_INFO is also fixed.

Thanks again for seeing these errors and help making our docs better.

closed 2018-06-26 by DOMARS