MicrosoftDocs/windows-driver-docs #560: Explanation lacks sufficient specificity to ensure proper implementation

BenWestbrook, 2018-06-09:
This one really puzzles me and causes me to wonder if I am missing something fundamental, if so, my apologies.

"...For an explanation of these values, see the SCSI-3 specification..."

Which one (document version, there are many)? Which section?

If these values are used by the OS, then a proper explanation of the requirements should be on this page including state machines and psuedo-code for each value (e.g. SRB_SIMPLE_TAG_REQUEST, etc.) reflecting the expectations of the OS.

Looking at sample and existing drivers, these values are largely ignored resulting in asynchronous completion of IRPs with mostly indeterminant order.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 32ffba3f-70cc-e42c-e0fe-bfb584b8a8eb
* Version Independent ID: 44ed080b-e7f2-cbc0-fb4b-9e260abd593c
* Content: [Storport's Interface with Storport Miniport Drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/storage/storport-s-interface-with-storport-miniport-drivers)
* Content Source: [windows-driver-docs-pr/storage/storport-s-interface-with-storport-miniport-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/storage/storport-s-interface-with-storport-miniport-drivers.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2019-05-03:
@lorihollasch, please add this one to your queue. I missed reassigning it when the writer of record left the team.

lorihollasch, 2019-06-19:
@BenWestbrook , I was finally able to update this page with the details that were missing. Thanks for reporting this, and apologies for the slow response!

closed 2019-06-19 by lorihollasch