MicrosoftDocs/windows-driver-docs #562: PDB do not contain enough information to decode ETL

gqsu, 2018-06-11:
SensorsCx.pdb and SensorsUtilsV2.pdb download from symbol server do not contain enough information to decode c88b592b-6090-480f-a839-ca2434de5844 related ETL logs.  Do you have any suggestion?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2cfec00c-8859-cca9-5f39-3ca65a09c852
* Version Independent ID: 55a1692b-ae3e-c16d-55be-9fbda035ebae
* Content: [Collecting and decoding WPP logs](https://docs.microsoft.com/en-us/windows-hardware/drivers/sensors/collecting-and-decoding-wpp-logs)
* Content Source: [windows-driver-docs-pr/sensors/collecting-and-decoding-wpp-logs.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/sensors/collecting-and-decoding-wpp-logs.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-06-11:
Thanks, @gqsu 
@mimisasouvanh can you research this. @DOMARS may be a good source for help.

JocelynBerrendonner, 2018-07-20:
Thank you for your suggestion! We opened an item in our bug system to capture your request. We will consider your request for future releases. At this point of time, we unfortunately can’t tell if and when these trace format symbols will be available.

Best regards,
Jocelyn Berrendonner [MSFT]

closed 2018-08-29 by mimisasouvanh