MicrosoftDocs/windows-driver-docs #563: What do you mean  "Provisioning is not supported for virtual machines."

adc68, 2018-06-11:
What happens if we really want the target to be a VM, like for instance because we want to be able to roll it back if the driver blue screens.  Surely this is the most useful and commonly desired setup, so to say it's not supported and give no reasons, nor tell us what happens if we try is really unhelpful.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ce050ea3-d61e-830c-3038-f725eb4c29e1
* Version Independent ID: eb7fc65e-bcfe-dc86-3f8c-336d20eb0c86
* Content: [Provision a computer for driver deployment and testing (WDK 10)](https://docs.microsoft.com/en-us/windows-hardware/drivers/gettingstarted/provision-a-target-computer-wdk-8-1)
* Content Source: [windows-driver-docs-pr/gettingstarted/provision-a-target-computer-wdk-8-1.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/gettingstarted/provision-a-target-computer-wdk-8-1.md)
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-06-11:
@DOMARS can you please clarify the passage?

duncanmacmichael, 2018-06-18:
Hi @adc68 , thanks for the question. We'll talk to the product owners and see if we can get the right answer for you about VM provisioning support.

gpotter2, 2018-06-27:
It sure would be nice to be able to test drivers on VMs...

jeh-cmkrnl, 2018-06-28:
You can certainly test drivers on VMs. I do it all the time, using a named pipe as a serial port for the debugger connection. What isn't supported is the automatic configuration for debugging, driver deployment, etc., as the dialog box indicates.

duncanmacmichael, 2018-07-31:
Hi @adc68 , we've gotten an answer back from the WDK feature team who confirmed that provisioning a VM *through the WDK* is not supported. This echoes what @jeh-cmkrnl mentioned. I'll update the doc for this change. Thanks again for your question and your patience.

closed 2018-07-31 by duncanmacmichael