MicrosoftDocs/windows-driver-docs #565: Not clear how notes re Windows 10 and 64 bit windows version interact

adc68, 2018-06-11:
Are the requirements for Windows 10 ("Starting with Windows 10, version 1607") in addition to the requirements for "64-bit versions of Windows starting with Windows Vista" or instead of them?  It would be great to get some clarity on the differences if any between signing or submission requirements for actual hardware drivers and virtual drivers such as filter drivers / LWF drivers etc.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a7cbefde-440f-b38e-8092-8a8a565fafe7
* Version Independent ID: cd915fda-1d67-ff23-4931-c1bf93705a5d
* Content: [Kernel-Mode Code Signing Requirements](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/kernel-mode-code-signing-requirements--windows-vista-and-later-)
* Content Source: [windows-driver-docs-pr/install/kernel-mode-code-signing-requirements--windows-vista-and-later-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/kernel-mode-code-signing-requirements--windows-vista-and-later-.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2018-06-12:
Hi @adc68, pretty sure the Win10 requirements are in addition to the 64-bit Vista and later requirements, but adding @SillyKeith to double-check.  I'm also not aware of any signing/submission differences for hw drivers vs virtual drivers, but again I'll default to Keith on that.

tedhudek, 2018-06-13:
Thanks for bringing this up @adc68.  I've made some updates to https://docs.microsoft.com/en-us/windows-hardware/drivers/install/kernel-mode-code-signing-requirements--windows-vista-and-later-, hopefully it's clearer now!

closed 2018-06-13 by tedhudek

adc68, 2018-06-13:
> <windows-driver-docs@noreply.github.com>
> Windows 10 and 64 bit windows version interact (#565)

Thanks Ted

So I guess this means from Windows 10 1607 onwards kernel drivers have 
several signatures required?  That's where my question came from.  The 
Attestation signing is therefore an additional signing step over and 
above the catalog or driver SPC signing.

Adrien