MicrosoftDocs/windows-driver-docs #566: Missing entry for 0x61949

LarryK348, 2018-06-11:
Received value of 0x61949 in bugcheck. This value is undocumented, what does it mean?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: de955911-dfbf-33b3-778a-767a5a578a0d
* Version Independent ID: 23fb5cbd-5f7d-f7ca-1622-5885020966c2
* Content: [Bug Check 0x1A MEMORY_MANAGEMENT](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x1a--memory-management)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2018-06-25:
Planning to consolidate the bug check 0x1A memory issues and get with the dev team this week.

DOMARS, 2018-06-29:
@LarryK348  - Information on the 0x61949 parameter is now included in the topic. Thanks for the input on the stop code topics.

closed 2018-06-29 by DOMARS