MicrosoftDocs/windows-driver-docs #572: Troubleshooting and resolution

LuigiBruno, 2018-06-12:
Please add troubleshooting and resolution steps for this bug check.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1e7f0759-3f68-e198-ef71-84d9dd55b259
* Version Independent ID: fe7c1046-56c8-e840-23cb-8d568c50dfbc
* Content: [Bug Check 0x12 TRAP_CAUSE_UNKNOWN](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x12--trap-cause-unknown)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x12--trap-cause-unknown.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x12--trap-cause-unknown.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

EliotSeattle, 2018-06-12:
Thanks, @LuigiBruno 
@DOMARS can you look into this, please?

DOMARS, 2018-06-27:
@LuigiBruno  - I added some resolution guidance on this stop code.  I also reviewed and corrected the parameter descriptions. Thanks for providing this input.

closed 2018-06-27 by DOMARS