MicrosoftDocs/windows-driver-docs #575: EDID for missing display

ADQTRW, 2018-06-13:
Is there a way to prevent Windows from rearranging or removing a display when it is disconnected?  NVIDIA drivers have this feature.  Other video chipset drivers from Intel do not.  Dependent upon the monitor, how it is connected (some DisplayPort to HDMI adapters), when the monitor is turned off, Windows removes and rearranges the displays.  Generally, not a good thing.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 55e79e97-9c23-a7a4-f7e9-748525022e9c
* Version Independent ID: ee067739-1ae0-4bd8-a206-4d36c31f17aa
* Content: [Overriding Monitor EDIDs with an INF](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/overriding-monitor-edids)
* Content Source: [windows-driver-docs-pr/display/overriding-monitor-edids.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/overriding-monitor-edids.md)
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2018-06-13:
@ADQTRW you might have better luck asking this on [Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
or [ntdev forum](http://www.osronline.com/showlists.cfm?list=ntdev), but assigning to @mimisasouvanh in case she knows

mimisasouvanh, 2018-06-18:
Yes, this is by design. When a display is disconnected, Windows will remove and rearrange the display to the active monitor.

closed 2018-06-18 by mimisasouvanh