MicrosoftDocs/windows-driver-docs #590: How to I get the latest product list?

shoebillk, 2018-06-19:
I need the recently submission item list. For that, I must get all product. 
How to I get the latest product list? Is therer a option for sort?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 40f09988-3ba3-0ce2-9952-b11ddd66aff1
* Version Independent ID: dd0d280c-f6a0-4a07-7f86-42eafaf6087b
* Content: [Get all products](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-all-products#feedback)
* Content Source: [windows-driver-docs-pr/dashboard/get-all-products.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/get-all-products.md)
* Product: **windows-hardware**
* GitHub Login: @balapv
* Microsoft Alias: **balapv**

EliotSeattle, 2018-06-19:
Thanks, @shoebillk for the question. @aahill can you please get this info into the topic, if it is possible?

aahill, 2018-06-20:
Hi @shoebillk, the API currently doesn't offer the ability to sort the products list, but is being considered for a future implementation. Any sorting will have to be done after you receive the whole product list for now.

closed 2018-06-20 by aahill

shoebillk, 2018-07-10:
Thanks, @aahill. I hope that sorting fuction wil be added in the near future.