MicrosoftDocs/windows-driver-docs #592: This topic lacks sufficient detail to properly implement.

BenWestbrook, 2018-06-19:
The reference to the SBC3 specification for thin and/or resource provisioning of a drive is insufficient guidance for proper implementation of this feature in a driver for Windows. For example, a driver that conforms to the above mentioned standard receives UNMAP SCSI requests from the OS for all unused blocks on the device. This occurs with DiskManager with "quick format" selected and with StorageSpaces. For a 4EB drive this will take years to complete. Our suspicion is there is something specific the OS is looking for that would prevent this scenerio. This would be a great place to expose that little nugget of information.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b67b2f1c-add1-0e22-29d1-8fe933af13c9
* Version Independent ID: 0ae1ca0f-68c6-5324-277d-d9d141eba1d5
* Content: [Thin Provisioning](https://docs.microsoft.com/en-us/windows-hardware/drivers/storage/thin-provisioning)
* Content Source: [windows-driver-docs-pr/storage/thin-provisioning.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/storage/thin-provisioning.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-06-19:
Thanks for the request, @BenWestbrook 
@aahill can you please tap the devs to make sure we have accurate info for this? Thanks!

EliotSeattle, 2019-05-03:
@lorihollasch, this Issue somehow fell through the cracks. Can you please put this in your queue?

lorihollasch, 2019-06-27:
@benwestbrook, the storage dev team indicated that thin provisioning is done by the Microsoft-provided disk driver, and is something that 3rd party drivers should not have to worry about. I will close this for now. If you still have questions and can provide more context, I will contact the team again - let me know. Apologies for the slow response time!

closed 2019-06-27 by lorihollasch