MicrosoftDocs/windows-driver-docs #594: More information about Message GUID

aventadorm, 2018-06-20:
I was expecting to have more information about how the message GUID is calculated.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 186def44-6dc1-37d4-55a8-df4e3008ab5f
* Version Independent ID: 63b036c2-f96d-64d2-4675-fe12135d485d
* Content: [Trace Message Format File](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/trace-message-format-file)
* Content Source: [windows-driver-docs-pr/devtest/trace-message-format-file.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/trace-message-format-file.md)
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-06-22:
Hi @aventadorm 
I'm not sure I exactly understand what you're looking for, but if you see the topics [Trace Message Header File](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/trace-message-header-file) and [WPP_CONTROL_GUIDS](https://msdn.microsoft.com/library/windows/hardware/ff556186) it may give the insight you need. If I'm off base, reopen this and let me know. Thanks, Eliot

closed 2018-06-22 by EliotSeattle

aventadorm, 2018-06-25:
I am trying to collect tmf files from builds and I am finding multiple duplicates in the same build. Checking the contents of the tmf files its either : 1) Same pdb files in different folders and same trace messages : or 2) Same pdb files in different folders but slightly different trace messages. If I had better insight or control over how these TMF files are named, I could better design a folder structure for the collection I am working on. Ideally, unique names for each tmf would be the best case for me since I could just dump them all together and optimize for large number of files in a folder. Since that is not possible, I was trying to better understand how the naming of these tmf files works. I run tracepdb on pdb files to generate these files. I had a look at both the links you provided and I am not quite able to grasp them but I will try going through them again tonight. Meanwhile, if there are any other references that suit my specific case, please let me know. I appreciate you quickly responding to my issue. Thank you !