MicrosoftDocs/windows-driver-docs #605: Display issues on page Bob Deinterlacing Algorithm

Powerhelmsman, 2018-06-25:
This issue was created by @drmandarin from https://github.com/MicrosoftDocs/feedback/issues/453
On this page:
https://docs.microsoft.com/en-us/windows-hardware/drivers/display/bob-deinterlacing-algorithm
Multiple display issues in formulae in the text, e.g:
0 &lt;= i &lt;= Mâˆ’1
0 &lt;= j &lt;=Nâˆ’1
(9*(b+c)âˆ’(a+d)+8)&gt;&gt;4
Please fix these issues.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b462c0f4-2d56-05a4-cc3c-c071a609590b
* Version Independent ID: 19e783f6-82c2-50c5-c344-01924ab9ea90
* Content: [Bob Deinterlacing Algorithm](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/bob-deinterlacing-algorithm)
* Content Source: [windows-driver-docs-pr/display/bob-deinterlacing-algorithm.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/bob-deinterlacing-algorithm.md)
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-06-25:
Thanks for raising this, @Powerhelmsman 
@mimisasouvanh can you please see if this is correct or needs editing? It looks the same as the WDCML version.

drmandarin, 2018-06-25:
No problem. I couldn't attach a screenshot to the original message, but
here's one for your reference so that you can clearly see an example of the
issue:

[image: image.png]

mimisasouvanh, 2018-07-23:
This has been fixed. Thank you for bringing this up!

closed 2018-07-23 by mimisasouvanh

drmandarin, 2018-07-23:
No problem and thanks for fixing.

Please check the correction of this equation:
(9*(b+c)âˆ’(a+d)+8)>>4

I have raised it because, since e.g. Mâˆ’1 was corrected to M-1, I would
expect that the fragment " âˆ’ " should be corrected to " - ".
Hence I would have expected that
(9*(b+c)âˆ’(a+d)+8)>>4

would be corrected to
(9*(b+c)-(a+d)+8)>>4

but this is not the case. Please review. Perhaps it was a typo on the
original incorrect page, so it isn't clear what the equation should be.

mimisasouvanh, 2018-10-09:
I went ahead and made this fix. Thank you so much!