MicrosoftDocs/windows-driver-docs #608: For driver-load-order dependencies, see How To Control Device Driver Load Order.

Shyamal-V, 2018-06-26:
"For driver-load-order dependencies, see How To Control Device Driver Load Order."

Link is broken

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 89caeb22-0006-6d94-24dc-b08bef273905
* Version Independent ID: 030be807-fb37-503f-f62a-1df6d19303c0
* Content: [Device management namespace objects](https://docs.microsoft.com/en-us/windows-hardware/drivers/bringup/device-management-namespace-objects)
* Content Source: [windows-driver-docs-pr/bringup/device-management-namespace-objects.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/bringup/device-management-namespace-objects.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-06-26:
Thanks for pointing this out, @Shyamal-V 
@barrygolden can you look into this? I think the KB article is gone, but you might find that install/specifying-driver-load-order is a valid resource.

barrygolden, 2018-06-26:
@Shyamal-V the link has been updated to point to [Specifying Driver Load Order](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/specifying-driver-load-order).

closed 2018-06-26 by barrygolden