MicrosoftDocs/windows-driver-docs #611: Typos my dude.

kwilliams1987, 2018-06-27:
or drives that are not available becuase of hardware regeneration

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 76528be9-25fa-14da-0f12-98e650278efa
* Version Independent ID: 22ae5b42-6beb-81b5-7652-1085a3bb92e6
* Content: [Bug Check 0x6B PROCESS1_INITIALIZATION_FAILED](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x6b--process1-initialization-failed)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x6b--process1-initialization-failed.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x6b--process1-initialization-failed.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

EliotSeattle, 2018-06-27:
Thanks, @kwilliams1987

DOMARS, 2018-06-27:
@kwilliams1987 - Thanks for pointing the this out. Fixes were made and an updated topic is now live.

closed 2018-06-27 by DOMARS