MicrosoftDocs/windows-driver-docs #613: missing /sr information

ticktickMOF, 2018-06-27:
missing information on /sr

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1f802ab4-103b-40bd-7aff-9b519b2e9f56
* Version Independent ID: 0d3ae05a-41f8-7cea-c9a8-25937b235905
* Content: [SymChk Command-Line Options](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/symchk-command-line-options)
* Content Source: [windows-driver-docs-pr/debugger/symchk-command-line-options.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/symchk-command-line-options.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

EliotSeattle, 2018-06-27:
Thanks, @ticktickMOF 
@DOMARS can you please look into this missing command?

DOMARS, 2018-06-29:
@ticktickMOF - The /sr option is no longer missing. Thanks for noticing this and letting us know.

closed 2018-06-29 by DOMARS