MicrosoftDocs/windows-driver-docs #618: Documentation is incorrect

LarryK348, 2018-07-01:
There is no "Creating UMDF-based HID Minidrivers"

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d4ed427e-a01a-368b-7422-2dea0aec50a3
* Version Independent ID: 1730981a-bdca-ba26-b8a1-301473678bb2
* Content: [Minidrivers and the HID class driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/minidriver-operations)
* Content Source: [windows-driver-docs-pr/hid/minidriver-operations.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/hid/minidriver-operations.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

aahill, 2018-07-02:
Hi @LarryK348, are you looking for the information contained on this page? htps://docs.microsoft.com//windows-hardware/drivers/wdf/creating-umdf-hid-minidrivers 

This topic should help with creating UMDF or KMDF Minidrivers using the WDF (Windows Driver Framework). The topic you've linked to is more focused on the HID Class Driver. I'll add the above link to this page to make it easier to find this information. Thanks!

closed 2018-07-02 by aahill

LarryK348, 2018-07-03:
(Windows Driver Model) WDM isn’t really applicable to User Mode (UMDF) drivers. (Windows Driver Foundation) WDF is.
I was specifically looking for information about UMDF HID minidrivers.
I expected the page (https://docs.microsoft.com//windows-hardware/drivers/wdf/creating-umdf-hid-minidrivers) to provide some useful information about UMDH HID minidrivers : what header files are used, and when; what are the correct libraries to link to; etc.

I believe there is a line on the original page like “Creating UMDF HID minidriver…”, but it doesn’t include a link and search results turn up nothing useful.

Larry

[Description: ac logo]
marcom, creative services, web & software development
Larry Kimminau
408.371.3962 x201
408.371.3823 (fax)
http://www.analyticconcepts.com<http://www.analyticconcepts.com/>