MicrosoftDocs/windows-driver-docs #621: Insufficient Guidance for Users Unable to View Win 10 WindowsUpdate.log

MikhailCompo, 2018-07-02:
Some Windows 10 users will not be able to read the WindowsUpdate.log due to missing symbols.
The vast *vast* majority of corporate customers machines are behind an internet proxy server, rendering the direct connect to download symbols totally unrealistic.
At the very least, Microsoft should be providing a command line example for how to create a manifest, because it is a PowerShell command and therefore the example above is not relevant and does not give sufficient information for how to debug *any* OS functions, and is solely aimed at someone who is debugging a single process they have created.
Please supply clear guidance/examples for how to create a manifest for the PowerShell command Get-WindowsUpdateLog in order to generate a usable log file when experiencing the (No Format Information found) message in the Windows Update log.
Thanks.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 69787f12-55d7-305a-35df-d005992763f4
* Version Independent ID: f5c8481d-a6e4-1aa9-901d-a2ee9e6fa278
* Content: [Using a Manifest File with SymChk](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/using-a-manifest-file-with-symchk#feedback)
* Content Source: [windows-driver-docs-pr/debugger/using-a-manifest-file-with-symchk.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/using-a-manifest-file-with-symchk.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

EliotSeattle, 2018-07-02:
Thanks for the suggestion, @MikhailCompo 
@DOMARS can please you look into what we can add to the docs and pass this on to the debugger team if there's an engineering request?

DOMARS, 2018-08-06:
@MikhailCompo - The dev team has received this information and is looking at options such as to removing the need for symbols with the [Get-WindowsUpdateLog](https://docs.microsoft.com/en-us/powershell/module/windowsupdate/get-windowsupdatelog?view=win10-ps) command.

In the meantime, I worked with the dev team to layout a few current options for managing symbols in an offline system when working with Windows update ETL logs in this topic.

[Offline Symbols for Windows Update](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/symbols-windows-update)

One important note covered in the docs is that the on and offline systems, must be running the same version of Windows - use winver or ver to confirm this. 

It is recommend using a single online machine so that the WU symbol cache will build month-by-month and contain the WU symbols from multiple update releases to completely parse the current update logs.

Thanks again for providing this input.

closed 2018-08-06 by DOMARS