MicrosoftDocs/windows-driver-docs #623: Might be missing PKEY_SensorData_Illuminance_Lux definition

githubtayomo, 2018-07-02:
The definition of "PKEY_SensorData_Illuminance_Lux" for Light sensor data fields may be "PKEY_SensorData_LightLevel_Lux".
Would you check this constant name right?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 215a7a8a-3b5c-f2f4-8c25-818fdd69455a
* Version Independent ID: 5e1d2143-0dc3-680b-a37f-da7e40e524ef
* Content: [Required data field properties](https://docs.microsoft.com/en-us/windows-hardware/drivers/sensors/required-data-field-properties)
* Content Source: [windows-driver-docs-pr/sensors/required-data-field-properties.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/sensors/required-data-field-properties.md)
* Product: **windows-hardware**
* GitHub Login: @mimisasouvanh
* Microsoft Alias: **windowsdriverdev**

JocelynBerrendonner, 2018-07-20:
Hello,

Thank you for reporting this issue! You are correct, the documentation has a typo. PKEY_SensorData_LightLevel_Lux is the correct value name. PKEY_SensorData_Illuminance_Lux doesn't exist.

Best regards,
Jocelyn Berrendonner [MSFT]

mimisasouvanh, 2018-07-20:
Issue fixed by PR: https://github.com/MicrosoftDocs/windows-driver-docs/pull/675

closed 2018-07-20 by mimisasouvanh