MicrosoftDocs/windows-driver-docs #624: Document markup artefact?

skycommand, 2018-07-03:
There is a problem with this page, as well as all its subpages ("Overview of the Boot.ini File", "Editing the Boot.ini File" and "Backing Up the Boot.ini File") and several other pages ("Boot Options in EFI NVRAM" and "Overview of Boot Options in Windows") in this topic: The phrase "[!IMPORTANT]" appears before one of the paragraphs at the top. This is, of course, far from good writership, but it suspect it is a markup element that didn't get rendered during the export to Microsoft Docs. The "[!NOTE]" markup seems to render perfectly in Microsoft Docs, forming a cyan note box, with an icon. In another page, "BCDEdit /set", I can see a "[!Caution]" markup. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 878e393f-198a-6d44-978d-9896aa10455b
* Version Independent ID: e8b27656-fdf3-0a69-f40c-2908c0744acb
* Content: [Boot Options in a Boot.ini File](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/boot-options-in-a-boot-ini-file)
* Content Source: [windows-driver-docs-pr/devtest/boot-options-in-a-boot-ini-file.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/boot-options-in-a-boot-ini-file.md)
* Product: **windows-hardware**
* GitHub Login: @nabazan-microsoft
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-07-03:
Thanks, @skycommand 
@barrygolden can you please fix these and look into whether we have a, gulp, global issue from migration?

barrygolden, 2018-07-03:
@skycommand the malformed alert text instances you found have been fixed. We'll scan the driver docs for other occurrences of malformed alert text and script a fix as required. Thanks for filing the issue.

closed 2018-07-03 by barrygolden

EliotSeattle, 2018-07-05:
I think you're on safe ground saying that you contributed to the Microsoft Windows Drivers documentation. :)
Thanks so much!

skycommand, 2018-07-05:
Oh! That's not what I meant. It was a joke...

Social decorum recommends resorting to a joke to counter tension or fear (in this case that of there being a global migration issue with far-reaching consequences) by making the affected people smile. I am not so petty as to tell anyone something like "I'm working with Microsoft" just because I wrote a comment. (Bah! Writing a comment! Kids do that! A lot!) Demonstrating such extreme degrees of hubris, I assumed, counts as joke. (Note to self: It was a failed joke.)

EliotSeattle, 2018-07-05:
We don't joke here. 
Just kidding. :-)